StefanBruens/ESP8266_new_pwm #3: undefined reference to pwm_init

mbostic, 2016-09-21:
I get this error:

```
Compiling 'pwmTest' for 'NodeMCU 1.0 (ESP-12E Module)'
pwmTest.cpp.o: (.text.setup+0xc): undefined reference to pwm_init(unsigned int, unsigned int*, unsigned int, unsigned int (*) [3])
pwmTest.cpp.o: (.text.setup+0x10): undefined reference to pwm_start()
 
pwmTest.cpp.o: In function setup
pwmTest.ino:23: undefined reference to pwm_init(unsigned int, unsigned int*, unsigned int, unsigned int (*) [3])
pwmTest.ino:24: undefined reference to pwm_start()
 
collect2.exe*: error: ld returned 1 exit status

Error compiling for board NodeMCU 1.0 (ESP-12E Module)

```

pwmTest.ino:

```
#include<pwm.h>

#define PWM_CHANNELS 1
const uint32_t period = 5000; // * 200ns ^= 1 kHz

							  // PWM setup
uint32 io_info[PWM_CHANNELS][3] = {
	// MUX, FUNC, PIN
	{ PERIPHS_IO_MUX_MTDI_U,  FUNC_GPIO12, 12 }
};

// initial duty: all off
uint32 pwm_duty_init[PWM_CHANNELS] = { 0 };

void setup() {
	pwm_init(period, pwm_duty_init, PWM_CHANNELS, io_info);
	pwm_start();
}

void loop() {
  
}
```

I have found "core_esp8266_wiring_pwm.c" in Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266 and replaced its contents with your code.

What am I doing wrong?

StefanBruens, 2016-09-21:
Your makefile is incorrect. Remove any "-lpwm" statement and add "pwm.o" instead.

This is not specific to this code, please read up about makefiles and linker statements.

closed 2016-09-21 by StefanBruens

mbostic, 2016-09-21:
@StefanBruens I have found this line in makefile: lpwm ?= ../../tools/xtensa-lx106-elf/bin/ so what should i change it to?

mbostic, 2016-09-22:
@StefanBruens Do I have to install complete esp8266 sdk to use this file?