StefanBruens/ESP8266_new_pwm #5: Exact replacement of libpwm.a

diffstorm, 2016-10-19:
Hi,

Could you share it as closed library named libpwm.a like exact replacement of the original one?

I'm using Arduino version of esp8266 software and I don't know how to add the pwm.c file to the project and where to configure it for just 3 pwm pins for rgb led controller.

Thanks in advance

nepeee, 2016-10-19:
1. Copy the pwm.h file from the arduino esp sdk directory to your arduino project directory.
(C:\Users\[your username]\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\tools\sdk\include\pwm.h)
2. Download and copy the pwm.c from this git to your arduino project directory.
3. Include the pwm.h in your project ino like this:
#include "Arduino.h"
extern "C" {
  #include "pwmm.h"
}

4. initialize the pwm in the setup function (in this example we use GPIO12-14):
  pinMode(12, OUTPUT);
  pinMode(13, OUTPUT);
  pinMode(14, OUTPUT);

  uint32 pwm_duty_init[3] = {0, 0, 0}; //initial duty cycle values 0-1000
  uint32 io_info[3][3] = {
    {PERIPHS_IO_MUX_MTDI_U, FUNC_GPIO12, 12},
    {PERIPHS_IO_MUX_MTCK_U, FUNC_GPIO13, 13},
    {PERIPHS_IO_MUX_MTMS_U, FUNC_GPIO14, 14}
  };

  pwm_init(1000, pwm_duty_init, 4, io_info);
  pwm_start();

5. Somewhere in your code update the pwm values like this:
 pwm_set_duty(100, 0); //GPIO12
 pwm_set_duty(500, 1); //GPIO13
 pwm_set_duty(800, 2); //GPIO14
 pwm_start();

diffstorm, 2016-10-24:
Thank you @nepeee for your help. I have implemented it as you pointed and it works.

I have 3 questions;

* I have added 255 as period. Is it good?

* Can I change the following definition's value to 3 from pwm.h file?
#define PWM_CHANNEL_NUM_MAX 8
Note: I observed that the pwm.c file does not include the pwm.h from my project. So when I remove the PWM_CHANNEL_NUM_MAX definition from pwm.c file, it warns me that it is not declared.
There is a bit confusing.

* Should I use PWM_USE_NMI 1 for better performance? It seems has no issue without it but if it is better, I will switch to it.

nepeee, 2016-10-26:
> I have added 255 as period. Is it good?

The 255 period value is very low but it could work until your code not using to much cpu time. See the other issue that i opened for details. For rgb leds i recommend a value between 20000-40000.

> Can I change the following definition's value to 3 from pwm.h file?
Note: I observed that the pwm.c file does not include the pwm.h from my project. So when I remove the PWM_CHANNEL_NUM_MAX definition from pwm.c file, it warns me that it is not declared.
There is a bit confusing.

Yes, i also changed the value, the lib uses less memory that way.
Simply define it in the ino file before the include.
#define PWM_MAX_CHANNELS 3
The pwm.c and the pwm.h uses different name for this value but i don't think it's a big problem.

> Should I use PWM_USE_NMI 1 for better performance? It seems has no issue without it but if it is better, I will switch to it.

The NMI interrupt has higher priority than the default timer interrupt, in my project it reduced the jitter of the output signals. For driving rgb leds i don't think its worth to switch to that mode if the effect of the jitter is not visible.

fernandoruizarg, 2017-09-19:
Hi, how can I configure more than 3 PWM channel ?
In all cases the setup would be ... I don't know what I would have to put here, for example in the case of GPIO16....

uint32 pwm_duty_init[4] = {0, 0, 0, 0}; //initial duty cycle values 0-1000
uint32 io_info[4][4] = {
{PERIPHS_IO_MUX_MTDI_U, FUNC_GPIO12, 12},
{PERIPHS_IO_MUX_MTCK_U, FUNC_GPIO13, 13},
{PERIPHS_IO_MUX_MTMS_U, FUNC_GPIO14, 14}
{???????????????????????????, FUNC_GPIO16, 16}
};

Thanks and sorry for my bad english

StefanBruens, 2017-09-19:
See https://lurchi.wordpress.com/2016/06/29/esp8266-pwm-revisited-and-reimplemented/#comment-21

fernandoruizarg, 2017-09-22:
Great job, thanks Stefan ! ! Works perfectly!
Another question, how can I change the frequency of the channels?

bigFin, 2017-11-03:
The info about pin registers, as well as including the pwm.h library in the Arduino project folder, aught to be added to the ReadMe.

bigFin, 2017-11-03:
On another note, this library works well with the NodeMCU v3. Thanks, Stephan!

closed 2018-03-19 by diffstorm