StefanBruens/ESP8266_new_pwm #11: exceptions when using delay

civic9, 2017-01-09:
I am trying to use this library with arduino.
Here is my code:

`
// PWM https://github.com/StefanBruens/ESP8266_new_pwm
#include "Arduino.h"
extern "C" {
#include "pwm.h"
}

#define PWM_CHANNELS 1
const uint32_t period = 5000;

void setup() {
  Serial.begin(115200);

  uint32 io_info[PWM_CHANNELS][3] = {
      {PERIPHS_IO_MUX_MTDI_U,  FUNC_GPIO12, 12}
  };
  
  uint32 pwm_duty_init[PWM_CHANNELS] = {0};

  pinMode(12, OUTPUT);

  pwm_init(period, pwm_duty_init, PWM_CHANNELS, io_info);
  pwm_set_duty(4, 0); //GPIO12 = D6
  pwm_start();
}

void loop() {
  delay(1);
}
`

Just setup and run simple loop, without even changing duty cycle.
It results with exception and reset every time, about 10 seconds after start.
Without delay(1) in the loop it works fine. It also works fine with other instructions in the loop - e.g. Serial.println(...);

Any ideas? Am I doing something wrong?

civic9, 2017-01-11:
Not sure if it is right solution, but tagging pwm_intr_handler with ICACHE_RAM_ATTR solves above issue for me.

Btw, maybe you consider changing licence to LGPL so it can be included in arduino libraries?

StefanBruens, 2017-01-11:
ICACHE_RAM_ATTR is a Arduino specific deviation from the SDK defaults, fixed nevertheless.

closed 2017-01-11 by StefanBruens