StefanBruens/ESP8266_new_pwm #12: Is it compatible with Arduino SDK?

Denages, 2017-02-17:
I'm trying to run example code, but have no results. PWM not working. 
`
#include "pwm.c"

#define PWM_CHANNELS 5
#define SDK_PWM_PERIOD_COMPAT_MODE 1
const uint32_t period = 5000; // * 200ns ^= 1 kHz
uint32 pwm_duty_init[PWM_CHANNELS] = {500, 500, 500, 500, 500};

uint32 io_info[PWM_CHANNELS][3] = {
    // MUX, FUNC, PIN
    {PERIPHS_IO_MUX_MTDI_U,  FUNC_GPIO12, 12},
    {PERIPHS_IO_MUX_MTDO_U,  FUNC_GPIO15, 15},
    {PERIPHS_IO_MUX_MTCK_U,  FUNC_GPIO13, 13},
    {PERIPHS_IO_MUX_MTMS_U,  FUNC_GPIO14, 14},
    {PERIPHS_IO_MUX_GPIO5_U, FUNC_GPIO5 ,  5},
};

 
 
void setup() {
  pwm_init(period, pwm_duty_init, PWM_CHANNELS, io_info);
  pwm_start();
}
 
void loop() {
}
`

fgather, 2017-02-28:
You have to set the respective pins to output mode. 
E.g. : 

`pinMode(15,OUTPUT); `

for GPIO15

StefanBruens, 2017-02-28:
Another Arduino stupidity.
GPIO directions and mode are set by pwm_init(...), but Arduino knows better ...

closed 2017-02-28 by StefanBruens