StefanBruens/ESP8266_new_pwm #16: Is a License change possible?

devyte, 2017-10-20:
There has been discussion on the ESP8266 Arduino repo about replacing/reworking the pwm code, i.e.: analogWrite().  One possibility that has come up is to use the code in this repo. However, the license is incompatible, as this code is GPL 2.0, while the Arduino repo is LGPL 2.1
Is it possible to change the repo license to the same as the ESP8266 Arduino repo for compatibility, so that this code could be used there?

jeffsf, 2017-11-29:
If amenable to it, a dual-license with BSD, MIT, Apache, or the like would be greatly appreciated. GPL, _especially_ LGPL, makes use in many situations impossible.

StefanBruens, 2017-11-29:
I would be interested in actual use cases not being able to comply with the GPL:

1. Hobbyists not willing to share some private hack used exclusively by themselves? Although sharing is appreciated, there is no need - you only have to provide the source if you give the device (and thus the running firmware) to someone else.

2. Commercial entities not willing to share their code? Really? You consider your code so valuable you are not willing to share, but happily leech other peoples code?

Demanding a BSD or other copyleft license makes it clear you are not willing to contribute. Telling LGPL restricts usage is completely wrong.

This code is the result of many man-hours spent. It is based on several years of education and professional experience, which don't come for free, nor do the hardware tools. I am happy to share on equal terms, but your request for a copyleft license tells me you are only willing to take, not to give back.

flannelhead, 2017-12-01:
@StefanBruens: I fully appreciate your decision to use a copyleft license such as GPL. (I think in the comment above you're mixing up copyleft and non-copyleft).

I once wished too that this project would be licensed under a MIT-compatible license. The background for this is a project of mine that's using the MIT-licensed [esp-open-rtos](https://github.com/SuperHouse/esp-open-rtos) as its base. I also use your PWM library which I've ported to work with esp-open-rtos. I thought it would be neat to be able to offer the port to the esp-open-rtos project which is currently missing multichannel PWM. I think this is one additional case to add to your list above, and perhaps a more compelling reason to change the licensing than the ones you've listed.

However, I realized that your choice of a copyleft license is fully understandable and that anyone wishing to use your code (or a derivative like mine) can release their own code under a compatible license. Thus I'm not requesting you to change the licensing.

There's one thing that is still not completely clear to me, though. Namely, there are those binary blobs from Espressif that one must link against to build a functioning binary for the ESP8266. These are licensed under the [Espressif MIT license](https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/License). If I understand correctly, the reason for LGPL licensing in the Arduino environment is the ability to link against these binaries. I'm not a lawyer, though, so I'm not sure if the same linking restriction applies for this project (or [mine](https://github.com/flannelhead/espway)) too.

StefanBruens, 2017-12-01:
@flannelhead: Regarding copyleft, you are right, when I wrote copyleft, I meant "lax" or "permissive".

Now, the GPL is compatible with permissive licenses, e.g. the Espressif license, see e.g. https://www.gnu.org/licenses/license-list.en.html#X11License - note, there is no "MIT license".

Linking GPLed software with software under different licenses is completely fine (https://www.gnu.org/licenses/gpl-faq.html#LinkingWithGPL), as long as all part are under compatible licenses.

Although Espressif calls their license MIT, the SDK code is not permissive licensed - they fail to provide the sources. They grant the right to modify the SDK software, but this right can not be exercised in a practical way.

Due to Espressif not providing the SDK under an appropriate license (although they claim something else), it is currently not possible to distribute a firmware in binary form. Using the SDK and the PWM library in a combined work is fine, as long as you do not distribute it.

flannelhead, 2017-12-01:
@StefanBruens: Thank you so much for clarifying this for me! Your explanation does make sense indeed.

Espressif sure aren't making things very easy for us open source developers. If I understand it correctly, I _could_ distribute your code and the SDK libs in binary form but they should reside in separate binary files, licensed under different licenses. Needless to say, this is pretty cumbersome. It's no fault of yours but rather Espressif's. I wish the situation with those blobs could be improved, but even ESP32's WiFi libs seem to be distributed only as binares, thus making no improvement.

But again, thank you for these clarifications!

flannelhead, 2017-12-09:
@StefanBruens, sorry to bother you again, but I would like to know what is your opinion on the request in the original post in this issue - switching to LGPL. You clearly indicated that using a permissive license such as BSD or X11 is not okay (and I agree - that would probably decrease the motivation to give improvements back to the community).

However, LGPL still ensures that modified source code is released. It however also makes it much more practical to work with the Espressif SDK. I, for instance, would probably one day like to sell or give away some copies of my robot running on ESP8266 and using your PWM library, while retaining the robot's source code licensed under a suitable license, most likely LGPL. That's not possible with the current GPL license and will probably lead me either to figure out a way to distribute the GPL parts separately (difficult if not impossible) or write a software PWM library of my own. LGPL would make it possible to use your library.

I also know that the Arduino ESP8266 core would benefit from this implementation as stated in OP.

devyte, 2017-12-13:
@StefanBruens  LGPL exists separate from GPL for various reasons. In the case of the Arduino/esp8266 repo, there are several contributors who work for companies, and most companies won't use GPL licensed code. That doesn't mean they don't contribute, it only means that they don't want to give up other pieces of code developed internally. This is usually due to competitive reasons, especially in small startups.
If a firmware were implemented that uses the code in this repo, GPL forces the entire thing to be GPL licensed, overriding the LGPL, and then those contributors couldn't use it.
Hence my request: is it possible to change this repo to LGPL,  or would you provide a special dispensation that an integration of this code into the Arduino/esp8266 repo be licensed under LGPL?

StefanBruens, 2017-12-14:
@devyte As can clearly be seen e.g. in case of the Linux kernel, the GPL motivates companies to give back. Companies are clearly able to use and contribute to GPLed code.

If you think you have keep code closed to be competitive, thats fine. But please do not demand from me to give up the competitive advantage offered by using the GPL. If you want to take advantage of this code, you are free to do, but only if you are willing to give back.

Seeing the high demand for using this library in apparently closed projects leads me to the conclusion this is something the other involved parties were not able to come up with on their own. You obviously see a high value in it - but you still rank your closed code so much higher?

devyte, 2017-12-14:
@StefanBruens the Linux kernel is not the same situation, as it is considered not part of any application. Hence, applications that rely on it are not required to be licensed in the same way.

The problem with closed source is that it gives back nothing. This is detrimental to open source code.
The problem with GPL is that it forces to give back everything, including trade secrets and technology that may be licensed from 3rd parties. This is detrimental to companies who can provide professional paid developers to the open source community.
LGPL and the like allow a middle ground.

Like I said, there are contributors from companies that do give back, In fact, if our repo were to change to GPL, we would lose most of our developers, and the repo would likely lose viability, at least for the immediate future.
I work for a company with both open and closed components, and I personally also have both open and closed code, which means I have both points of view. I am not demanding anything from you, nor am I looking to get into a debate about which license is better, or how things should be done. If you developed this code on your own, without basis on anything else, you are free to license it any way you want. But as things stand, you are free to use our code under your terms if you wished (your work using ours would be GPL), but we can't use your code under our terms. And since we can't use it, we also can't contribute to it, such as we have e.g.: with umm_malloc most recently, where we just found a fundamental bug and its fix.

I am looking for a middle ground. If you don't wish to change the license in your repo, a special dispensation under LGPL only for Arduino/esp8266 would be a compromise. Consider that the Arduino/esp8266 repo is not closed source. It is licensed under LGPL, but is openly hosted, and the source code is freely available.

I do not think that all code needs to be closed to remain competitive. I also don't think that all code must be open to allow an open source community. Say you want to use the ESP8266 to implement a wifi bridge to a 3rd party serial device that uses a proprietary serial protocol. The Arduino/ESP8266 code allows this, but if your code were used in addition, then it would not be possible due to the implementation of the protocol. If your code were LGPL, the firmware source could be released to the community, and the implementation of the protocol could remain closed, e.g.: in binary library form, thereby meeting all requirements.

Our repo already has a pwm implementation. However, from comments in our repo, I suspect yours is better, although at this time I don't know for sure. I also suspect I could develop something similar to what you have here, given enough time. But why spend significant effort reinventing the wheel? Isn't cooperation one of the pillars of open source? We use and cooperate with umm_malloc, SPIFFS, lwip, and others. We're inviting you to be part of it.

Are you open to this?

ianmacs, 2018-01-20:
To summarize:

- ESP8266/Arduino wants to integrate StefanBruens/ESP8266_new_pwm.
- ESP8266/Arduino is LGPL and does apparently not want to change their own license.
- StefanBruens/ESP8266_new_pwm is GPL and is asked here to change license from GPL to LGPL.
- StefanBruens says no.

So why is this still open? Please close.

Whoever is interested in making the integration happen, can
- fork ESP8266/Arduino
- integrate StefanBruens/ESP8266_new_pwm
- The combined work can now be distributed under GPL
- Applications using combined work are also GPL

Where is the problem?

devyte, 2018-01-20:
We can't change our license for the same reason I explained above: there are products that make use of our code, and also of closed licensed code, so GPL is a no go.
The problem with what you said is that it's not an integrated solution: each person will need to do it, or if it's put in a repo, the integration will need to be maintained by someone. Even if a maintainer steps forward, which is unlikely only for a new pwm implementation, it means the code base will eventually diverge.
I was waiting for a final reply, but I guess there is no point. Closing.

closed 2018-01-20 by devyte