StefanBruens/ESP8266_new_pwm #17: pwm at 25Khz

fireport, 2017-10-27:
Hi, i should want to drive a 3 wire pc fan with this library. Wich is the better way to set pwm to the 25Khz need to smooth operations of the fan ? Thank you

StefanBruens, 2017-12-10:
Please clarify your question.

fireport, 2017-12-12:
I solved the problem using this code

`#define PWM_PERIOD 125000 
// PWM channels

#define PWM_CHANNELS 3

// PWM setup (choice all pins that you use PWM)

uint32 io_info[PWM_CHANNELS][3] = {
  // MUX, FUNC, PIN
  //  {PERIPHS_IO_MUX_GPIO5_U, FUNC_GPIO5,   5}, // D1
  //{PERIPHS_IO_MUX_GPIO4_U, FUNC_GPIO4,   4}, // D2
  //{PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0,   0}, // D3
  // {PERIPHS_IO_MUX_GPIO2_U, FUNC_GPIO2,   2}, // D4
  //{PERIPHS_IO_MUX_MTMS_U,  FUNC_GPIO14, 14}, // D5
  {PERIPHS_IO_MUX_MTDI_U,  FUNC_GPIO12, 12}, // D6
  {PERIPHS_IO_MUX_MTCK_U,  FUNC_GPIO13, 13}, // D7
  {PERIPHS_IO_MUX_MTDO_U,  FUNC_GPIO15 , 15}, // D8
  // D0 - not have PWM :-(
};`

Thanks

2bedom, 2018-05-26:
hi there,

i tried to run a 4 wire pwm fan which needs 25khz pwm signal... without success..

- i use a wemos D1 mini wired  GOPIO 15 // D8 to the fan green wire(pwm)
- vcc and gnd for the fan comes from a seperate source.

the code i tried was this
`
#include "Arduino.h"
extern "C" {
#include "pwm.h"
}

void setup() {
  #define PWM_CHANNELS 1
  const uint32_t period = 125000; // * 200ns ^= 25 kHz

// PWM setup
uint32 io_info[PWM_CHANNELS][3] = {
  // MUX, FUNC, PIN 
  {PERIPHS_IO_MUX_MTDO_U,  FUNC_GPIO15, 15},
};
//pinMode(15, OUTPUT);

// initial duty: all off
uint32 pwm_duty_init[PWM_CHANNELS] = {0, 0, 0, 0, 0};

pwm_init(period, pwm_duty_init, PWM_CHANNELS, io_info);
pwm_start();
Serial.begin(115200);
}

void loop() {
  pwm_set_duty(70000, 0);  // GPIO15: 10%
  pwm_start();
  Serial.println("");
  Serial.print(pwm_get_duty(1));
  Serial.print("###1 ");  
  delay(5000);
  pwm_set_duty(125000, 0);  // GPIO15: 100%
  pwm_start();
  Serial.println("");
  Serial.print(pwm_get_duty(1));
  Serial.print("###2 ");
  delay(5000);
  pwm_set_duty(0, 0);  // GPIO15: 0%
  pwm_start();
  Serial.println("");
  Serial.print(pwm_get_duty(1));
  Serial.print("###3 ");
  delay(10000);
}
`

i copierd the pwm.h from the arduino folder to my project folder, also the pwm.c from the git repo
after that i opened them both in arduino with ->file -> open  (sry iam new on arduino normaly i run lua code on the devices..)

result: fan didnt care about the set duty cycles .. doesnt matter what i set... the only thing i recognized was, if a wire up the pin to pwm it reduces the speed a little bit... buts thats it...

any ideas... what did i miss ?!

Thanks

gordonthree, 2018-07-15:
hi @2bedom any luck sorting your problem out? I think the period needs to be smaller, not bigger. eg the example states a period of 256 (8 bit res) for 19khz. So maybe try 128 but maybe that would be too fast? I am about to experiment with this as well, i'll post results soon as I get the hardware together. My fans are 24v and on an isolated ground, so I need to find an opto coupler to drive the pwm input with a 3.3v signal.

2bedom, 2018-07-15:
@gordonthree yes i was .. i am using the wemos d1 mini , all worked fine i had 2 things to deal with... one was my own stupidity.

1.) my fan had a pwm range  from 300rpm-1700rpm so setting the duty to zero will never stop him :)
2.) common ground - i missed that i had to use the same ground !

so now iam using a noctua pwm fan, common ground and everything worked fine .. 
if you need some further infos or code snipped feel free to ask..

StefanBruens, 2018-07-15:
25kHz is 1/25000 seconds, i.e. 40 microseconds or 40000 nanoseconds. Thus the correct period for 25kHz is:

period * ticktime = 40000
=>
period = 40000 / 200 = 200

2bedom, 2018-07-15:
@StefanBruens  hmm seems to be clear ... but why does my 125000 work Oo ?

i setthe duty in 10k steps from 10k to 125k ... 

so if i switch to 200 period i'll have to scale/map 0-100% from 0-200 right ?

xuhongv, 2018-08-02:
hi，in the rtos v3.0 SDK，when i add the pwm.c , build find error：ETS_FRC_TIMER1_INTR_ATTACH()  undefine ..