StefanBruens/ESP8266_new_pwm #18: Only works at duty cycles of 0% and 100%

xoseperez, 2017-12-05:
I'm experiencing an issue with some RGB controllers. I'm using low frequencies (period = 10000). Using NMI works great but when I use TIMER1 only duty cycles of 0 or 10000 (100%) work. Anything in between just doesn't change anything. I have the feeling it could be hardware related (maybe the SPI flash?). Any ideas?

StefanBruens, 2017-12-10:
Each hardware timer can only be used by one functional block. Obviously something else grabs the TIMER1 interrupt, so the PWM outputs are constant.

closed 2017-12-10 by StefanBruens

MayaPosch, 2017-12-14:
I'm having a similar issue currently, with neither the TIMER1 or NMI version working at all. Any attempt to set the duty with the period set to 1 kHz (compatibility mode enabled) results in an output voltage of either 0 or 3.3V on the selected pin.

asturel, 2017-12-14:
Same issue, but with NMI it works fine, I did a grep for 'ETS_FRC_TIMER1_INTR_ATTACH' in my libraries, but nothing came up, also the esp8266's pwm implementation use NMI too (https://github.com/esp8266/Arduino/blob/master/cores/esp8266/core_esp8266_wiring_pwm.c#L162)

MayaPosch, 2017-12-14:
I think I got it to use NMI, but it's hard to check what the resulting binary is using. The PWM_DEBUG flag seemed to be working at least.

StefanBruens, 2017-12-14:
The complete low level interrupt handling is hidden in the on-chip ROM, so this almost impossible to debug. You can not query which interrupt handler is currently set for a specific interrupt, nothing.

Unfortunately, all the missing documentation adds up here - Espressif is not documented their execution model, the Arduino Esp8266 HAL does not really tell which timer interrupts it is using for its own needs ...

You may have some luck examining the firmware with e.g. objdump or gdb, it should show all call sites calling either ets_isr_attach or _xt_isr_attach.