StefanBruens/ESP8266_new_pwm #19: Migrate to C++ class?

nixmeer, 2017-12-10:
Hello :)

I am working on a board that can easily control up to 5 pwm outputs or 1 RGB light an 2 pwm outputs using one ESP12f. It is all up and running using the FadeLed class (https://github.com/septillion-git/FadeLed) which is based on the analogWrite(). In order to drive led (former halogen) bulbs, I need a low active option for my board. Besides that, I can hear some harmonics of the 1 kHz PWM. I tried to use your class including it using 

```
extern "C" { 
    #include <pwm.h> 
}
```
but it wouldn't work. The error I get is 

> /Users/jt/Documents/Arduino/libraries/FadeLedNewPwm/src/FadeLedNewPwm.cpp:32:9: error: expected constructor, destructor, or type conversion before '(' token
 pwm_init(period, pwm_duty_init, PWM_CHANNELS, io_info);
         ^
/Users/jt/Documents/Arduino/libraries/FadeLedNewPwm/src/FadeLedNewPwm.cpp:33:12: error: expected constructor, destructor, or type conversion before ';' token
 pwm_start();
            ^
/Users/jt/Documents/Arduino/libraries/FadeLedNewPwm/src/FadeLedNewPwm.cpp:34:1: error: expected declaration before '}' token
 }

I guess the reason for this is that the compiler expects a "FadeLed::" in front of every function, right? 

I tried to understand your code in order to implement it in to the FadeLed class but failed. Do you have any idea or could change your code to an independent class?

I study electronics at THM in Gießen, GER, so I try my best in programming but I'm not into it very deeply. If communicating with me is easier in German, feel free to contact me :-) 

Best

StefanBruens, 2017-12-10:
How about providing actual source code? The error messages without context are useless ...

nixmeer, 2017-12-10:
You're right. Give me one moment to check things again before asking for your time ...

nixmeer, 2017-12-10:
The header file of the FadeLed looks like this:

```
/**
 *  @file FadeLed.h
 *  @brief A simple Arduino library to fade LEDs on the hardware PWM.
 *  
 *  This library takes all the heavy lifting of fading a LED. 
 */

#define PWM_CHANNELS 1

#ifndef _FADE_LED_H
#define _FADE_LED_H

#if defined(ARDUINO) && ARDUINO >= 100
#include "Arduino.h"
#else
#include "WProgram.h"
#endif

#if defined ( ESP8266 )
  #include <pgmspace.h>
#else
  #include <avr/pgmspace.h>
#endif

/**
 *  @brief Sets the number of bits used for the PWM.
 *  
 *  Set this to a higher value for devices with higher PWM resolution. Will set the upper limit and variable type correspondingly.
 *  
 *  **Default** 8-bit: 0-255 for Uno/Nano/Pro Mini etc
 *  
 *  @warning Can't simply increase the number to have more PWM levels. It's limited to the hardware.
 */
 #if defined ( ESP8266 )
  //ESP8266 has 10-bit PWM
  #define FADE_LED_PWM_BITS 10
#else
  //Change this to match the number of PWM bit on other devices
  #define FADE_LED_PWM_BITS 8
#endif

/**
 *  @brief Sets the variable type used for the brightness.
 *  
 *  Is done automatically according to #FADE_LED_PWM_BITS. A byte (uint8_t) is used for 8-bit or less. An unsigned int (uint16_t) is used for anything above. So limited to 16-bit PWM.
 *  
 *  @see FADE_LED_PWM_BITS
 */
#if FADE_LED_PWM_BITS <= 8
typedef uint8_t flvar_t;
#else
typedef uint16_t flvar_t;
#endif

/**
 *  @brief Maximum number of FadeLed objects
 *  
 *  **Default** = 6, the number of hardware PWM pins on an Uno/Pro Mini/Nano
 */
#ifndef FADE_LED_MAX_LED
#define FADE_LED_MAX_LED  6
#endif

/**
 *  @brief The maximum brightness step of the PWM
 *  
 *  Automatically set depending on #FADE_LED_PWM_BITS. 255 for 8-bit.
[... 717 lines cut ...]
    io_info[0][1] = FUNC_GPIO11;
    io_info[0][2] = 11;
  } 
  else if (pin == 12) {
    io_info[0][0] = PERIPHS_IO_MUX_MTDI_U;
    io_info[0][1] = FUNC_GPIO12;
    io_info[0][2] = 12;
  } 
  else if (pin == 13) {
    io_info[0][0] = PERIPHS_IO_MUX_MTCK_U;
    io_info[0][1] = FUNC_GPIO13;
    io_info[0][2] = 13;
  } 
  else if (pin == 14) {
    io_info[0][0] = PERIPHS_IO_MUX_MTMS_U;
    io_info[0][1] = FUNC_GPIO14;
    io_info[0][2] = 14;
  } 
  else if (pin == 15) {
    io_info[0][0] = PERIPHS_IO_MUX_MTDO_U;
    io_info[0][1] = FUNC_GPIO15;
    io_info[0][2] = 15;
  }
}
```

nixmeer, 2017-12-10:
or do you prefer the files?

nixmeer, 2017-12-16:
Hello again :) 

I created a separate class that can control GPIO2 and GPIO10 but the other pins won't work. I'm not sure if it is possible to control pin 5 (GPIO16) that way but digitalWrite() is working. Pin 2-4 are not working as well but I can't find the reason why.

this is the header:

```ruby
#define PWM_CHANNELS 6
#include "Arduino.h"

class AachenerPwm
{
public:
	AachenerPwm(uint32_t period);
	static void setPwm(int _onTime, byte _pin, bool _start);
private:
	uint32_t io_info[PWM_CHANNELS][3]; 
	uint32_t pwm_duty_init[PWM_CHANNELS];
	uint32_t period;
};
```

the cpp files is:
```ruby
#include "/Users/jt/Documents/Arduino/libraries/AachenerPwm/AachenerPwm.h"
extern "C" { 
    #include </Users/jt/Documents/Arduino/libraries/AachenerPwm/pwm.h> 
}

AachenerPwm::AachenerPwm(uint32_t _period)
{
	period = _period;

	pwm_duty_init[0] = period;
	for(int i=1; i<PWM_CHANNELS; i++){
	    pwm_duty_init[i] = 0;
	}

	io_info[0][0] = PERIPHS_IO_MUX_GPIO2_U;
	io_info[0][1] = FUNC_GPIO2;
	io_info[0][2] = 2;

	io_info[1][0] = PERIPHS_IO_MUX_SD_DATA3_U;
	io_info[1][1] = FUNC_GPIO10;
	io_info[1][2] = 10;

	io_info[2][0] = PERIPHS_IO_MUX_MTCK_U;
	io_info[2][1] = FUNC_GPIO13;
	io_info[2][2] = 13;

	io_info[3][0] = PERIPHS_IO_MUX_MTDI_U;
	io_info[3][1] = FUNC_GPIO12;
	io_info[3][2] = 12;

	io_info[4][0] = PERIPHS_IO_MUX_MTMS_U;
	io_info[4][1] = FUNC_GPIO14;
	io_info[4][2] = 14;

	io_info[5][0] = PERIPHS_IO_MUX + 0x44;
	io_info[5][1] = 1;
	io_info[5][2] = 16;

	pwm_init(period, pwm_duty_init, PWM_CHANNELS, io_info);
	pwm_start();
}

void AachenerPwm::setPwm(int _onTime, byte _pin, bool _start){
	pwm_set_duty(_onTime, _pin);
	if(_start){
		pwm_start();
	}
}
```

StefanBruens, 2018-03-17:
GPIO16 is in a different register location, so it will not work.
For the others, there may be some Arduino stuff overwriting the registers. Thats the reason I don't use Arduino, shoehorning AVR semantics into the ESP can only work to a certain degree.

Have you tried calling the functions directly?