StefanBruens/ESP8266_new_pwm #23: Multiple Definitions Error when Compiling

jlubins, 2018-03-20:
Hi,

I receive this error when attempting to compile mariusmotea's diyHue code (which uses your pwm.c as an addition):

```
Arduino: 1.8.4 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, QIO, 1M (128K SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Archiving built core (caching) in: /var/folders/yf/v0mm1gt17d75p1rdyd5vjlkw0000gn/T/arduino_cache_140262/core/core_esp8266_esp8266_generic_CpuFrequency_80,ResetMethod_ck,CrystalFreq_26,FlashFreq_40,FlashMode_qio,FlashSize_1M128,led_2,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_7f04165fe51592c926f91225eac66eae.a
sketch/Generic_RGBW_Light.ino.cpp.o: In function `pwm_start':
sketch/pwm.c:356: multiple definition of `pwm_start'
sketch/pwm.c.o:sketch/pwm.c:356: first defined here
sketch/Generic_RGBW_Light.ino.cpp.o: In function `_Function_base':
sketch/pwm.c:403: multiple definition of `pwm_set_duty'
sketch/pwm.c.o:sketch/pwm.c:403: first defined here
sketch/Generic_RGBW_Light.ino.cpp.o: In function `pwm_init':
sketch/pwm.c:415: multiple definition of `pwm_get_duty'
sketch/pwm.c.o:sketch/pwm.c:415: first defined here
sketch/Generic_RGBW_Light.ino.cpp.o: In function `pwm_init':
sketch/pwm.c:426: multiple definition of `pwm_set_period'
sketch/pwm.c.o:sketch/pwm.c:426: first defined here
sketch/Generic_RGBW_Light.ino.cpp.o: In function `String::operator==(char const*) const':
sketch/pwm.c:161: multiple definition of `pwm_init'
sketch/pwm.c.o:sketch/pwm.c:161: first defined here
sketch/Generic_RGBW_Light.ino.cpp.o: In function `operator()':
sketch/pwm.c:436: multiple definition of `pwm_get_period'
sketch/pwm.c.o:sketch/pwm.c:436: first defined here
sketch/Generic_RGBW_Light.ino.cpp.o: In function `operator()':
sketch/pwm.c:442: multiple definition of `get_pwm_version'
sketch/pwm.c.o:sketch/pwm.c:442: first defined here
sketch/Generic_RGBW_Light.ino.cpp.o: In function `operator()':
sketch/pwm.c:446: multiple definition of `set_pwm_debug_en'
sketch/pwm.c.o:sketch/pwm.c:446: first defined here
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Generic ESP8266 Module.
```

I reached out to mariusmotea about the issue and he directed me here. I would greatly appreciate if you could shed some light on why this is occurring. Does it have to do with a conflict with function definitions contained in the ESP8266 board manager? Thank you for your time.

StefanBruens, 2018-03-20:
Apparently the code has been copied into Generic_RGBW_Light.ino.cpp, and is also provided by pwm.c.

 Not a bug of this code.

closed 2018-03-20 by StefanBruens

jlubins, 2018-03-21:
I see. Thank you. There is no repeat definition in either file, so I think it is an issue with the way Arduino itself compiles the code.