StefanBruens/ESP8266_new_pwm #24: SDK_PWM_PERIOD_COMPAT_MODE logic incorrect

zym060050, 2018-04-02:
Hi,

The logic for the SDK_PWM_PERIOD_COMPAT_MODE appears to be incorrect?

#if SDK_PWM_PERIOD_COMPAT_MODE
#define PWM_PERIOD_TO_TICKS(x) (x * 0.2)
#define PWM_DUTY_TO_TICKS(x) (x * 5)
#define PWM_MAX_DUTY (PWM_MAX_TICKS * 0.2)
#define PWM_MAX_PERIOD (PWM_MAX_TICKS * 5)
#else
...
#endif

In README:

By default there is one small difference to the SDK. The code uses a unit of 200ns for both period and duty. E.g. for 10% duty cycle at 1kHz you need to specify a period value of 5000 and a duty cycle value of 500, a duty cycle of 5000 or above switches the channel to full on.

To have full compatibility with the SDK, you have to set the SDK_PWM_PERIOD_COMPAT_MODE define to 1. If set, the code will use 1us for PWM period and 40ns for the duty cycle. E.g. 10% duty cycle at 1kHz is set by a period value of 1000 and a duty cycle value of 2500, full duty at 25000 and above.

Let's say for 1KHZ, if SDK_PWM_PERIOD_COMPAT_MODE is defined to be 1. Then period should multiply by 5 (1000*5=5000) not divided by 5.

The correct logic should be:

#if SDK_PWM_PERIOD_COMPAT_MODE
#define PWM_PERIOD_TO_TICKS(x) (x * 5)
#define PWM_DUTY_TO_TICKS(x) (x * 0.2)
#define PWM_MAX_DUTY (PWM_MAX_TICKS * 5)
#define PWM_MAX_PERIOD (PWM_MAX_TICKS * 0.2)
#else
...
#endif

Thanks.

ge0rg, 2018-05-13:
I just came here to say the same thing. From my experiments a long time ago, I found out that the maximum duty at a period of 1000 on my H801 module is actually around 50505 (don't ask me why or how I found out, I think I re-calculated it from one of the Espressif SDK examples).

But even with 25000 it doesn't work without the change by @zym060050.

perpernorbi, 2018-08-23:
I have created a pull request (https://github.com/StefanBruens/ESP8266_new_pwm/pull/26), which will fix this issue.