StefanBruens/ESP8266_new_pwm #30: Flickering

zdila, 2019-10-15:
Hello,

First thanks for the library which I consider to be the best OSS PWM library for ESP8266 I found.

I am observing soft LED flickering when using this PWM implementation.

The complete code:

```c
#include "c_types.h"
#include "pwm.h"
#include "eagle_soc.h"

uint32 io_info[][3] = {
  { PERIPHS_IO_MUX_MTCK_U, FUNC_GPIO13, 13 }
};

uint32 pwm_duty_init[5] = {20};

void user_init() {
  pwm_init(250, pwm_duty_init, 1, io_info);
}

uint32_t user_rf_cal_sector_set() {

}
```

Measurements with oscilloscope shows that most of the pulses take 4.07 μs:

[image]

But if I set trigger to pulse width of higher values, then it often triggers:

[image]

Original firmware of the device, which is closed source, doesn't have this issue. But according to strings in the binary blob it seems to be build with RTOS.

Is there anything I can do to prevent this?

StefanBruens, 2019-10-15:
Thanks for the praise, let me return it - finally, a well done bug report ...

As the PWM is done in software, there is always the possibility for some other interrupt handler to delay the PWM interrupt handler. Are there any other interrupt routines?

One possible mitigation is to lower the PWM frequency, 20kHz is quite high for lighting applications.

Are the long pulses isolated, or do they appear in groups? Can you do a trace including several pulses?

zdila, 2019-10-15:
Thank for your prompt reply.

> Are there any other interrupt routines?

None I am aware of. Testcase only uses ESP8266_NONOS_SDK, your PWM implementation and the code from this report.

I also had to add this line as `ICACHE_RAM_ATTR` is not defined in ESP8266_NONOS_SDK:
```c
#define ICACHE_RAM_ATTR __attribute__((section(".text")))
```
I set it up using PlatformIO.

I am observing the same problem in my project based on Mongoose OS which internally uses ESP8266_NONOS_SDK 2.1.0.

> One possible mitigation is to lower the PWM frequency, 20kHz is quite high for lighting applications.

I've tried to set period to 1200 for the same duty (20) but the problem is still there.

> Are the long pulses isolated, or do they appear in groups? Can you do a trace including several pulses?

They seem to be isolated. I captured a trace with many pulses and focused on that wide one, then checked cca 10 on both sides and they were of "normal" width.

zdila, 2019-10-25:
As I mentioned our product uses Mongoose OS. After we added Azute IoT functionality, the PWM quality rapidly decreased.

Surprisingly (for me) `#define PWM_USE_NMI 1` fied the problem.